netlabtoolkit/VarSpeedServo #1: High-resolution servo support.

cevich, 2015-02-26:
Looking at the code, it seems the range is hard-coded to 0-180 and pulse-widths are limited to multiples of 4us.  However, modern digital/programmable servos allow programming the travel-range to increase the resolution (i.e. same pulse-widths, fewer degrees). It's not uncommon to have 0-120 or 0-90 servos.  Also, I think this suggests new overloaded``write()`` methods that accept degree-fractions (or radians), or centi-degrees.

What do you think are the challenges involved in removing the 4us multiple (for all servos) and allowing to specify angle constraints per-servo (or all servos)?

(I'm willing to help if you think the goal is possible)

netlabtoolkit, 2015-02-26:
Hi Chris,

Depending on the resolution of the Arduino hardware timers, I think it is probably possible.

It might be worth looking at the most recent standard Arduino Servo library to see if they’ve made any updates in the direction you are talking about. VarSpeedServo is based on the standard library, but was forked from it quite a while ago.

.phil

cevich, 2015-02-26:
Wow, fast reply!  Naw, IMHO latest upstream Arduino Servo library still includes a bunch of "suck" as it seems this code is loosely based on http://scolton.blogspot.com/2010/07/arduino-interrupt-driven-servo-routine.html I haven't actually looked, but just by using it, the upstream code seems to be blocking on write(), not running asynchronously :cry:  But that's fine, it's more of a general implementation whereas this one is more specialized and limited.

It just happens that I'm in need of higher resolution and speed control to achieve coordinated movements with two servos (i.e. so they both end moving different distances at the "same" time).  Otherwise my laser will go zig-zaggy across the floor and my cat won't have so much fun :grinning: 

I think making the angle-constraints more flexible will be the easy part, I'm not sure on how to tweak the ISRs and timers for better resolution.  Thanks for your input, and no rush, my cat is very patient.

netlabtoolkit, 2015-02-26:
> Wow, fast reply! Naw, IMHO latest upstream Arduino Servo library still includes a bunch of "suck" as it seems this code is loosely based on http://scolton.blogspot.com/2010/07/arduino-interrupt-driven-servo-routine.html <http://scolton.blogspot.com/2010/07/arduino-interrupt-driven-servo-routine.html> I haven't actually looked, but just by using it, the upstream code seems to be blocking on write(), not running asynchronously  But that's fine, it's more of a general implementation whereas this one is more specialized and limited.

I won’t have time for a few weeks, but if you want to make some code suggestions, that would get me started faster. ;)

.phil

cevich, 2015-02-26:
I'm in the same boat, limited time for tinkering.  I'm happy to do the legwork then send pull-requests with code, just not sure I fully understand the signal manipulations of these ISRs.  I've worked with avr timers, interrupts, and low-level IO before, it's just not all together at once :grin:   So code-stufy tips would be appreciated, then I'm happy to hack away at it.

cevich, 2015-03-11:
FWIW, I looked over the Arduino 1.5 code, the ISR does not block on high or low, so I was incorrect in my assertion of 'suck'.  However, it is commented that the minimum resolution is 4uS and they've fixed the pulse period at 20ms by a macro :cry: According to WikiPedia, many servo's can handle signaling much faster and at higher resolution than that, some as low as 6ms between pulses!

Anyway, since I need to synchronize my servo movements (i.e. so servoX and servoY arrive at the about same time), I needed to implement my own speed-control.  However I think the upstream Arduino servo library could be greatly enhanced by killing the fixed 4uS and 20ms values.  I'm wondering if you understand how/why there is 4uS resolution?  Does it have to do with the timer tick-duration?  If so, can't the tick-time just be reduced by >4uS and then drop the hard-coded /4 calculation?

When u have time, no rush.  Thanks!