netlabtoolkit/VarSpeedServo #9: Add support for ESP8266

marcelogaio, 2016-05-07:
Is there any chance this would work on an ESP8266?
Thanks!!!!!!!!!

joihn, 2017-02-06:
same problem, it use avr stuff which is not compatible with eps :/

SimonME30, 2017-02-09:
I was hopeful
https://github.com/Q1an/Project-Tau/blob/master/code/ESP8266/servo/servo.ino

 - incomplete project I think. In VarSpeedServo.cpp I commented out 
#include <avr/interrupt.h> as that is now std with the arduino IDE
Changed #include <WProgram.h> to #include <arduino.h>

It then fell apart and I ran out of talent very quickly

BobMcNair, 2018-04-10:
Yes this would be a very useful feature!
Shame it cant be done yet!

pvanallen, 2018-04-11:
As owner of this library, I agree. My version of VarSpeedServo is built on prior work that was started a long time ago. I think the only path is to rebuild the library on the modern servo libraries so VarSpeedServo isn't tied down to the AVR. For example, then we could then make use of:

https://github.com/esp8266/Arduino/tree/master/libraries/Servo

This would require coming up with a new architecture and rewriting much of the underlying code. I don't have time for that at this point, but perhaps in the summer. If anyone can find any other libraries out there that are more modern and could be adapted, that would be a big help. Or if anyone with low level servo experience wants to volunteer some time and to collaborate with me, that would be much appreciated!

dlarue, 2019-05-21:
Same here. I've been looking for a nice servo easing library and thought I found it with VarSpeedServo and tested on a Nano.  No-go on the NodeMCU which I'm using because of the WiFi access and date/time features.

Bozzibo, 2020-01-07:
Any update on this? :)

till234, 2020-11-10:
> As owner of this library, I agree. My version of VarSpeedServo is built on prior work that was started a long time ago. I think the only path is to rebuild the library on the modern servo libraries [...]

is ther any uptdate to this? does the library work with nano 33iot? or Nano pretzelboard? awesome library. would be even better to use it in iot projekts.. thx

pvanallen, 2022-05-03:
> Sorry, but it would be too much work to update this library for all the different hardware (I'd be happy to turn this repository over to somewhat can put in the effort though). You might check out my new library [VarSpeedPython](https://github.com/pvanallen/VarSpeedPython) which serves a similar purpose. Of course it is for Python platforms like CircuitPython and MicroPython which are supported on many different hardware platforms. VarSpeedPython supports timed transitions from one value to another with options for easing and the number of steps used in the transition. The move() and sequence() calls are designed to be called within an event loop, and do not block. Hope this helps!

dlarue, 2022-05-03:
@pvanallen, it seems to be a good move doing the circuitpython and micropython route over addressing each variant of esp.
It looks like the circuitpython folks have handled that part well with all the supported hardware( https://circuitpython.org/downloads ). 

 I need to get back to my Kinetic Clock project and this gives me an excuse for porting the Arduino code to circuitpython and giving your VarSpeedPython library a whirl.

navdotnetreqs, 2023-02-01:
Still hoping for this to be supported on esp :S

chatmurai, 2025-04-24:
It's 2025, esp8266 is still around here and I am struggling to find a library that can move a servo with speed parameter 😭