netlabtoolkit/VarSpeedServo #11: wait for the end of the movement of two servo?

jgrossrieder, 2017-01-08:
Hi,

I've two servo(basically mounted as a tower, one is for X one for Y). 

I've some new coordinates to go(like go to the angle 50° in X, 60 in Y).

The issue that I've is that if I only wait on the second instruction, maybe y has to go from 59 to 60, and x from 0 to 90, and in this case the movement is not finish when the next steps are solved, any idea how to solve this?

To wait on the end of the two servos, which will not have the same duration?

GillHawk, 2017-02-09:
Hi, a forked this project to add the wait feature (https://github.com/GillHawk/VarSpeedServo) and made a pull request to merge with the original project. 
I am not sure if this is what you want (two servos moving in parallel and wait for both to finish):
  servoX.write(posX, SPEED1);     
  servoY.write(posY, SPEED2);
  servoX.wait(); // wait for servo X to finish
  servoY.wait();  // wait for servo Y to finish
...
..process next command

gargakshit, 2017-07-08:
## You could use

ServoName.write(degrees, speed, true/false);

## True for the code execution to wait until servo motion is complete, false to continue the code execution even if servo is still running...

# Hope it helped...

jgrossrieder, 2017-07-08:
@gargakshit  This doesn't work. You want to wait that the two movements are complete, BUT execute them in the same time.

gargakshit, 2017-07-16:
@jgrossrieder Use true with second servo and false with first servo

Example:

servoOne.write(degrees, speed, false);
servoTwo.write(degrees, speed, true);

_Sent from my YU5010 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

jgrossrieder, 2017-07-16:
@gargakshit so you really think it would work if the first movement takes more time than the second one? I've often the case, and I need to be sure that both movements are finished (and I never know which one takes longer

GillHawk, 2017-07-16:
I needed the same function in the past. Wait for both servo to finish then do another move. This will look like this:

servoX.write(posX, SPEED1);
servoY.write(posY, SPEED2);
servoX.wait(); // wait for servo X to finish
servoY.wait(); // wait for servo Y to finish

..
The wait command forces a loop to wait for finish.  
If first wait is done, wait for second one (if is the case).

Added code changes here (seems that pool request was ignored):
https://github.com/GillHawk/VarSpeedServo

jgrossrieder, 2017-07-16:
@GillHawk Yes, that would be perfect, this should be merged back.

pvanallen, 2017-07-23:
@jgrossrieder @GillHawk I just merged this. Thanks for the contribution!

closed 2017-07-23 by pvanallen