netlabtoolkit/VarSpeedServo #14: Motor Reset when I use attach function.

gargakshit, 2017-05-15:
# My servo comes to 0 when I use attach function.
## Here is the code I used:

#include <VarSpeedServo.h>

VarSpeedServo test;

void setup() {
  // put your setup code here, to run onc
  test.attach(2);
}

void loop() {
  // put your main code here, to run repeatedly:
}

MatanRubin, 2017-05-26:
I'm seeing the same issue - servo resets position when calling `attach()`, instead of just staying where it is. Except for this issue this library is great - exactly what I was looking for - thanks!

jackdewhurst, 2017-07-05:
Same here, unfortunately this makes the library unusable for my project :-(.

gargakshit, 2017-07-05:
True. I, because of this library, lost a gold medal in robotics competition.

gargakshit, 2017-07-08:
__
# ISSUE__ 

### This problem also occurs with the standard Servo library, also this library is an extension of the Standard Servo library, so the developer might not fix it...

gargakshit, 2017-07-08:
Test

pvanallen, 2017-07-08:
Hi @gargakshit, thanks for clarifying, and offering a work-around. I'm the owner of the library. I will admit that I have not had time recently to work on this repo. I'm am looking for someone to help resolve some of the other open issues and pull requests.

gargakshit, 2017-07-08:
@pvanallen Sir, should I re-open the issue???

jackdewhurst, 2017-07-10:
@gargakshit this issue does not occur for me using the standard Servo library.

When I use **write()** directly after **attach()** it still moves to 0 before moving to the desired value, even when wait = false. This doesn't happen using the Servo library.

I've remedied it by exposing **DEFAULT_PULSE_WIDTH** from the library, reading the servo position then setting it before calling **attach()**.

gargakshit, 2017-07-15:
@jackdewhurst Hey, thanks for the update. This solved the issue. 

### Thanks Again

_Sent from my YU5010 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

pvanallen, 2017-07-23:
@jackdewhurst, @gargakshit  I've reopened this issue - Jack, could you submit a pull request and I'll integrate it? Or just post the changes. Thanks!

gargakshit, 2017-11-02:
**What's the status???**

pvanallen, 2017-11-02:
@gargakshit Could you post the code you used to resolve this?

cape, 2017-12-03:
Hi, I am having the same issue. 
when attach a servo, the servo moves to 0. I need to avoid this, thanks

sapics, 2019-10-11:
I create the PR to fix this issue.
I have not tested the PR yet, because I don't have servo motor.
It would be helpful, if you could test it.