netlabtoolkit/VarSpeedServo #19: isMoving can't return "false"

andreas-seiderer, 2018-01-23:
The "isMoving" function misses a return "false" if the conditions for "true" are not met.

"Flowing off the end of a value-returning function (except main) without a return statement is undefined behavior." [source](http://en.cppreference.com/w/cpp/language/return)

Please add "return false;" at the end of the function and it should work like expected.

```
bool VarSpeedServo::isMoving() {
  byte channel = this->servoIndex;
  int value = servos[channel].value;
  
  if (value < MIN_PULSE_WIDTH) {
    if (read() != value) {
      return true;
    }
  } else {
    if (readMicroseconds() != value) {
      return true;
    }
  }
  
  return false;
}
```

GillHawk, 2018-01-23:
Indeed, I missed that.. the shorter version would be:
```

bool VarSpeedServo::isMoving() {
  byte channel = this->servoIndex;
  int value = servos[channel].value;
  
  if (value < MIN_PULSE_WIDTH) {
    **return (read() != value);**
  } else {
    **return (readMicroseconds() != value);**
  }
}
```

closed 2018-01-23 by pvanallen

pvanallen, 2018-01-23:
I went with the original suggested fix for clarity. Thanks @andreas-seiderer

commit b626d86e (2018-01-23): fixed #19 accounting for false condition