netlabtoolkit/VarSpeedServo #22: Servo not waiting until move complete before executing another move

a-hodgson, 2018-08-04:
The code below sets up and initiates a move from 20 degrees to 170 degrees and then back to 20 degrees all at full speed waiting for each move to complete before starting the next:

`VarSpeedServo servo1;
servo1.attach(5);
servo1.write(20, 0);
servo1.wait();

servo1.write(170, 0);
servo1.wait();

servo1.write(20, 0);
servo1.wait();`

However the servo is not completing the move to 170 degrees before returning to 20 degrees.

GillHawk, 2018-08-11:
I did not test your exact scenario but I suggest that you should to first check if the servo can indeed move to 170 degrees (there are some servos that can reach less than 180 degrees).

Test this to see if it reaches the expected 170 degrees:
servo1.attach(5);
servo1.write(170, 0);

You can also test this using standard servo library,

GillHawk, 2019-04-20:
Hi, I know is too late now but today I was playing a bit with the lib and I noticed that using
"servo1.write(20, 0)" as you described, indeed it fails.
The solution now is to use full speed "255" instead of "0".
servo1.write(20, **255**);

GillHawk, 2019-04-20:
The problem was that method write(int value, uint8_t speed), for speed 0 calls write(int value) that sets the "tick" directly so wait() will return immediately.
I am working on a quick fix now.