netlabtoolkit/VarSpeedServo #33: Library update for Arduino Nano 33 IoT and 33 BLE?

Bjerg13, 2020-02-06:
Hallo,
The new Arduino Nano’s (Every, 33 IoT, 33 BLE and 33 BLE Sense) are using new MCU’s and are not compadeble with the current VarSpeedServo library. Will there be an update anytime soon?

Thanks

till234, 2020-11-10:
did anyone try the varspeesservo library with nano 33 iot yet? right know im using a nodemcu together with a arduino nano in order to slow down my servoapp.. however i got little jittering since im using puls out and in....

smappdooda, 2022-04-29:
Wondering the same thing. I just got 3 Nano Every's and get this error "'OCIE1A' was not declared in this scope". Are there any plans to make this compatible?

pvanallen, 2022-05-03:
Sorry, but it would be too much work to update this library for all the different hardware (I'd be happy to turn this repository over to somewhat can put in the effort though). You might check out my new library [VarSpeedPython](https://github.com/pvanallen/VarSpeedPython) which serves a similar purpose. Of course it is for Python platforms like CircuitPython and MicroPython which are supported on many different hardware platforms. VarSpeedPython supports timed transitions from one value to another with options for easing and the number of steps used in the transition. The move() and sequence() calls are designed to be called within an event loop, and do not block. Hope this helps!