netlabtoolkit/VarSpeedServo #39: Update for SAMD21

Mark-Kenworthy, 2021-04-07:
I need a version of this that works on the SAMD21 processor for a project of mine.  I'm working on an update for that processor and will contribute back when complete.

till234, 2021-04-08:
awesome

pvanallen, 2022-05-03:
> Sorry, but it would be too much work to update this library for all the different hardware (I'd be happy to turn this repository over to somewhat can put in the effort though). You might check out my new library [VarSpeedPython](https://github.com/pvanallen/VarSpeedPython) which serves a similar purpose. Of course it is for Python platforms like CircuitPython and MicroPython which are supported on many different hardware platforms. VarSpeedPython supports timed transitions from one value to another with options for easing and the number of steps used in the transition. The move() and sequence() calls are designed to be called within an event loop, and do not block. Hope this helps!